kipr/Simulator #6: Fix and test motor functions

tcorbly, 2020-05-11:
- [ ] clear_motor_position
- [ ] ao
- [ ] move_to_position

prithvirajkadiyala, 2020-05-13:
Do we clear_motor_position on reset?

prithvirajkadiyala, 2020-05-15:
Change the ao() to set motor power to 0 rather than setting the motor mode to 0.
Because the motor() call sets the motor mode to forward when called.

prithvirajkadiyala, 2020-05-17:
JavaScript processing the bitwise operations on base 16 values. We will need it to process base 2 values. ao() will work when we move the bases from 16 to 2

prithvirajkadiyala, 2020-05-18:
Fixed it by calling motor(port, 0) in libwallaby

prithvirajkadiyala, 2020-05-18:
Fix the clear_motor_position_counter by adding EMJS code to set registers to zeros in this line here.
https://github.com/kipr/libwallaby/blob/c9cac3c446bd02bcb422515e2986a956d62d749a/src/motors_c.cpp#L28

commit ede3ad92 (2020-05-20): Fixed #6
closed 2020-05-20 by prithvirajkadiyala