kipr/Simulator #81: Robot does not stop motors when not locally run

tcorbly, 2021-04-21:
Robot does not stop motors when not locally run. The program ending or an ao() does not appear to stop it either. This issue only appears when connecting to the server on AWS, not on locally run instances.

navzam, 2021-04-23:
I think PR #84 will partially fix this by supporting `ao()` as it exists in libwallaby `master`.

End-of-program behavior might still be broken. I noticed an emscripten error in the console. I can take a look at that.

tcorbly, 2021-04-23:
Yes, I can confirm PR#84 did fix this excluding end-of-program behavior.

navzam, 2021-04-24:
Cool, then I'll re-title the issue to be about end-of-program behavior.

I've traced this down to an emscripten issue. We had taken care of this in PR #24, when we were using emscripten 1.40.1 (or somewhere close to that). I confirmed that using the latest emscripten (2.0.18) causes the issue. Now trying to see if we can easily fix it without reverting emscripten. Otherwise we may need to revert until we figure it out.

navzam, 2021-04-27:
Here's what I've found out. Currently emscripten doesn't allow async functions (like `emscripten_sleep()`) to be called during shutdown, including static destructors. So the cleanup logic in `Wallaby::~Wallaby()` is crashing because it calls `ao()`, which eventually calls `Private::set_motor_mode()`, which calls `readRegister8b()`, which calls `emscripten_sleep()`.

For an immediate fix, I'd suggest reverting to emscripten v2.0.2. @tcorbly if you're comfortable with that, I can change the install script to install a specific emscripten version.

Longer term, there are a few options so that we aren't stuck on an old version of emscripten:
1. For the simulator specifically, run the cleanup logic a different way than in the destructor.
1. In libwallaby, remove the dependency on destructors and run the cleanup logic in a different way.
1. Wait for the emscripten issue to be fixed. I submitted it in emscripten#13993

navzam, 2022-09-04:
Closing this issue. Since PR #396 switched to `SharedArrayBuffer`, we no longer use async functions in libwallaby. PR #395 then switched to a much newer version of emscripten (v3.1.19)

closed 2022-09-04 by navzam