kipr/Simulator #87: Add more friction to wheels or mat for turning

tcorbly, 2021-04-23:
Robot spins out of control when just turning one wheel. Probably need to adjust friction between wheels and mat to prevent this.
Code for testing below:

`#include <stdio.h>
#include <kipr/wombat.h>

int main()
{
  printf("Hello, World!\n");
  motor(0, 100);
  msleep(2500);
  return 0;
}`

navzam, 2021-04-24:
I remember messing with the friction to fix slipping issues, and I found that increasing gravity (from `-9.8` to `-9.8 * 10` made a huge improvement. I wanted to go to the real-world value of `-9.8 * 100` but started getting weird physics behavior, which is why it's lower.

navzam, 2021-05-21:
@tcorbly do you think this is fixed now? I ran one wheel at full speed for 20 seconds and it seemed reasonable.

closed 2021-06-01 by tcorbly