kipr/Simulator #194: Verify Simulator works in all major browsers/OSes

bmcdorman, 2021-07-14:
I propose the following test matrix:

**OS Set**
- Ubuntu 20.04 LTS
- Windows 10
- macOS Big Sur

**Browser Set**
- Chrome 91
- Firefox 90
- Safari 13 (macOS only)

navzam, 2021-07-14:
For consistency, I suggest at least testing these things:
- Simulator area and UI elements all load properly
- UI widgets can be expanded/maximized/closed
- Expanding/collapsing the subsections in the Robot and World widgets
- Setting robot position/rotation in the Robot widget
- Enabling cans and changing surfaces in the World widget
- Running the default program compiles successfully and prints to the console
- Clicking "Stop" before the program finishes works. Consider adding some `motor()` commands and a long `msleep()`.
- Introducing a syntax error and trying to run results in a) console errors, b) inline editor errors, and c) error badge in editor header
- "Download" button works
- "Clear" button on console works
- "Layout" button displays the layout menu, and "Show All" / "Hide All" buttons work
- "Documentation" button opens docs in a new tab
- "About" button opens the About dialog
- Simulator framerate is "acceptable"

navzam, 2021-07-14:
I've been developing/testing with Edge (91.0.864.67) on Windows 10 and it's all good. I can try Edge 88 if that specific version is important.

navzam, 2021-07-15:
Tested with Chrome (91.0.4472.124) on Windows 10, all good.

navzam, 2021-07-15:
Tested with Firefox (90.0) on Windows 10. Everything works but the sim framerate is noticeably lower. I'll open a separate issue and we should profile at some point.

bmcdorman, 2021-07-28:
Marking this as resolved, as we haven't discovered any major issues. If anything arises please open an issue.

closed 2021-07-28 by bmcdorman