kipr/Simulator #201: Add an easy way for people to submit issues/feedback

navzam, 2021-07-16:
We should have some way for people to submit issues/feedback directly from the simulator, or at least instruct people on how to submit them externally. Some ideas:

- In the About dialog, direct people to the GitHub issues page if they have a GitHub account, or provide a KIPR email address otherwise.
- In a separate Feedback dialog, allow users to enter their feedback and optionally their email address. Our backend could then send an email to a KIPR email address, or directly post the issue on GitHub

will-jac, 2022-01-14:
Starting work on a dedicated Feedback dialog that lives in the main menu.

Eventually, the feedback might go to a database, where a microservice could automatically post it to a channel in discord

navzam, 2022-01-16:
Resolved by #292 and #295

closed 2022-01-16 by navzam