kipr/Simulator #342: Add controller IDE functionality

Zacharyprime, 2022-04-11:
Harrogate currently has a host of errors that appear when trying to build from a fresh RaspiOS.
The errors can probably be resolved without much issue, but the plan is to eventually switch to Ivygate/other anyway.

For that reason, it is worthwhile to experiment with the following.
1. Test if the simulator will run on a Wombat and what kind of performance can be obtained with it. If the performance is too poor to be usable, the simulator tab could be removed to accomplish 2 and 3 on hardware.
2. Add a new tab to the simulator which is essentially just the monaco code editor but with compile, save, etc buttons that build to Wombat/controller hardware like in harrogate (basically what Ivygate was supposed to be but built off the existing simulator code). 
3. Add a drone programming tab or incorporate drone programming into the IDE.

Right now we have a branch called "IDE" for these experiments to take place in.

I was hoping to package v27 with a new and/or improved IDE. v26.1 doesn't seem to have any new problems yet and v25.6 is stable enough to keep us afloat for a while, so we have some time to get v27 off on a good footing. 

The experimental OS instructions can be found here:
https://zacharyprime.github.io/KIPR-DevDocs/

navzam, 2022-04-24:
@Zacharyprime for 1 and 2, are you saying we'd run the simulator project in this repo on the controller directly? I assume the controller just needs the IDE portion (editor + buttons), so shouldn't the controller have its own project (the successor of harrogate) that depends on ivygate?

Zacharyprime, 2022-04-25:
@navzam Since the IDE has that box menu where you can select the different pages, I was going to add a separate option for IDE to that. 

We may end up making this a separate repository, but first I wanted to at least try running the simulator on a Wombat just to see if we can get that extra functionality added to the selling point of the controller. 

My plan was to make a separate repository if that experiment failed, but now that I'm rethinking this, it might seem confusing for a person trying to access our hosted simulator to see the IDE tab.

bmcdorman, 2022-04-26:
We should start a harrogate successor project based loosely on the simulator (UI elements in particular). I don't think we should target a Wombat simulator on the Wombat for now, though.

Zacharyprime, 2022-04-27:
Maybe there are aspects of the simulator I didn't realize.

I was intending to just remove all the login stuff from the simulator and make it so that you just type in the IP and immediately get access to both the simulator and the IDE from one menu.

I don't actually care about the simulator part of it, I just thought it would be a neat bonus since I was hoping to just reuse the simulator UI and strip out everything that wouldn't be useful in the editor (ie the game/physics engine).

Based on your response I'm guessing the code is structured differently than I was expecting. I hadn't gotten very far into this idea yet, all I did was add a new IDE page that is basically a copy of the simulator page.

closed 2023-02-17 by Zacharyprime