kipr/Simulator #345: Mobile height is incorrect

will-jac, 2022-04-19:
[100vh is not what you expect on mobile](https://chanind.github.io/javascript/2019/09/28/avoid-100vh-on-mobile-web.html) ([another link](https://allthingssmitty.com/2020/05/11/css-fix-for-100vh-in-mobile-webkit/))

This means that the site can scroll past the header, leading to weird behavior where the browser navigation bar obscures the simulation menu.

This occurs on both Safari and Chrome on iPad.

Possible solutions:
- Fix the root cause by using CSS
- Make the sim menu float on the page

I'm working on the first solution and hope to have a PR by this weekend

will-jac, 2022-04-19:
Another possible solution is to see if we can get the address bar to shrink by default. This would be the best option as it would give us more vertical space to work with on mobile

navzam, 2022-07-02:
Fixed in will-jac's PR above

closed 2022-07-02 by navzam