kipr/Simulator #378: Items sometimes shrink slightly while moving them around

navzam, 2022-07-01:
In rare cases, moving an item around using the gizmo can cause it to shrink slightly. It's hard to repro consistently, but you can try:

1. Add a paper ream
2. Click the ream to select it. Move/rotate the ream in various ways
3. Without unselecting the ream, rotate the camera to a different angle
4. Move the ream toward/away from you. This will sometimes result in the shrinking behavior

navzam, 2022-07-01:
The scale of the Babylon mesh seems to actually be changing. Nothing in our code should be changing the scale, so I'm thinking this is a Babylon bug. I'll try to confirm this

tcorbly, 2024-12-19:
Tried reconfirming this but haven't been able to reproduce. This may be fixed but further testing should be done.