kipr/Simulator #428: Youtube links no longer work

tcorbly, 2022-11-08:
Links to the tutorial videos are broken and may need to be updated.

chrismbirmingham, 2023-05-16:
I did a bit of digging on this but I don't yet have a solution. The problem is caused by the Cross-origin isolation required for using features like SharedArrayBuffer:

function setCrossOriginIsolationHeaders(res) {
  res.header("Cross-Origin-Opener-Policy", "same-origin");
  res.header("Cross-Origin-Embedder-Policy", "require-corp");
}

It would seem that [the current suggested solution](https://developer.mozilla.org/en-US/docs/Web/Security/IFrame_credentialless) is to use the credentialless tag in the iframe, but that is not currently available on even a majority of browsers...

I would suggest we drop the iframe embeddings and replace them with links, as right now people don't have a way to access the videos without directly searching for them on youtube.

tcorbly, 2024-12-19:
Resolved in #504

closed 2024-12-19 by tcorbly