kipr/Simulator #459: Localization for login page

navzam, 2023-02-07:
The login page is not localized because we don't know the user's locale until they've logged in. I think other sites handle this by having a locale/region picker in the corner and ideally defaulting to some preference from the browser.

tcorbly, 2024-12-19:
Would this also be something we could try detecting via the IP?

navzam, 2024-12-19:
@tcorbly I think IP has some problems, like multiple languages in a location, or users being behind proxies. We can try to use the [`Accept-Language` header](https://developer.mozilla.org/en-US/docs/Web/HTTP/Headers/Accept-Language) or [`navigator.language` property](https://developer.mozilla.org/en-US/docs/Web/API/Navigator/language) to pick a default. I think most important is to have a picker somewhere before the user logs in. Choosing a default is bonus